ghzserg/zmod #109: Ошибка пути

noyhay1, 2025-05-13:
Решил поставить сервер на orange pi zero 3 да телеграм бота.

Во первых в документации https://github.com/ghzserg/zmod/wiki/Macros_en#one-command-install-debian
Ссылается на bash <(wget --cache=off -qO- https://github.com/ghzserg/zmod_ff5m/raw/main/telegram/telegram.sh)  
Но такого пути нет.

При запуске https://github.com/ghzserg/zmod_ff5m/tree/1.5/telegram/telegram.sh

Выдаёт ошибку. Приложил скрин.

После создания бота, и если написать ему, он не отвечает, чтобы узнать chat_ID
Помогла эта инструкция https://sitogon.ru/blog/252-kak-sozdat-telegram-bot-poluchit-ego-token-i-chat-id 

Ещё Debian в orange pi подключены Huawei репозитории, вместо Debian, может с этим связано.

[image]
[image]

ghzserg, 2025-05-13:
Документация исправлена
Правильная команда
```
bash <(wget --cache=off -q -O - https://github.com/ghzserg/zmod_ff5m/raw/refs/heads/1.5/telegram/telegram.sh)
```

noyhay1, 2025-05-13:
По поводу получения chat_id добавьте пожалуйста пояснение. У меня бот не отвечает

Станет активным окно ввода текста. Необходимо написать что нибудь нашему боту. По возможности приятное.

Не ждите, что он ответит.

Однако, без сообщений боту мы не сможем получить ID чата с ним.

Итак, сообщение отправлено.

Теперь мы должны создать текст вида:

https://api.telegram.org/bot<Токен вашего бота>/getUpdates

Где вместо <Токен вашего бота> необходимо вписать  ранее полученный токен вашего бота.

Вставляем данный текст в адресную строку браузера.

Если после перехода в браузере вы увидели, что то вроде того, что на рисунке ниже и ничего более, попробуйте написать боту еще раз.

После чего возвращаемся в браузер и обновляем страницу.

Откроется что то вроде:

Нас интересует значение рядом с ID. Копируем его.

closed 2025-05-14 by ghzserg