ghzserg/zmod #110: ошибка развертывания

noyhay1, 2025-05-13:
Установлена debian 12 на orangepi zero 3
при запуске скрипта bash <(wget --cache=off -q -O - https://github.com/ghzserg/zmod_ff5m/raw/refs/heads/1.5/telegram/telegram.sh)

выходит ошибка 

dpkg: ошибка при обработке архива /var/cache/apt/archives/tzdata_2025b-0+deb11u1_all.deb (--unpack):
 не удалось открыть «/usr/share/zoneinfo/posix/Africa/Abidjan.dpkg-new»: Нет такого файла или каталога
При обработке следующих пакетов произошли ошибки:
 /var/cache/apt/archives/tzdata_2025b-0+deb11u1_all.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)
Следующие пакеты имеют неудовлетворённые зависимости:
 python3-pyrsistent : Зависит: python3 (< 3.10) но 3.11.2-1+b1 должен быть установлен
 python3-setuptools : Зависит: python3-pkg-resources (= 52.0.0-4+deb11u1) но 66.1.1-1 должен быть установлен
 python3-yaml : Зависит: python3 (< 3.10) но 3.11.2-1+b1 должен быть установлен
E: Невозможно исправить ошибки: у вас зафиксированы сломанные пакеты.

root@orangepizero3:~# bash <(wget --cache=off -q -O - https://github.com/ghzserg/zmod_ff5m/raw/refs/heads/1.5/telegram/telegram.sh)
Сущ:1 http://deb.debian.org/debian bullseye InRelease
Сущ:2 http://security.debian.org bullseye-security InRelease
Сущ:3 http://deb.debian.org/debian bullseye-updates InRelease
Сущ:4 http://deb.debian.org/debian bullseye-backports InRelease
Сущ:5 https://download.docker.com/linux/debian bookworm InRelease
Чтение списков пакетов… Готово
Построение дерева зависимостей… Готово
Чтение информации о состоянии… Готово
Может быть обновлён 1 пакет. Запустите «apt list --upgradable» для показа.
Чтение списков пакетов… Готово
Построение дерева зависимостей… Готово
Чтение информации о состоянии… Готово
Расчёт обновлений… Готово
Следующие пакеты будут обновлены:
  tzdata
Обновлено 1 пакетов, установлено 0 новых пакетов, для удаления отмечено 0 пакетов, и 0 пакетов не обновлено.
Необходимо скачать 0 B/307 kB архивов.
После данной операции объём занятого дискового пространства возрастёт на 941 kB.
Preconfiguring packages ...
(Чтение базы данных … на данный момент установлено 205311 файлов и каталогов.)
Подготовка к распаковке …/tzdata_2025b-0+deb11u1_all.deb …
Распаковывается tzdata (2025b-0+deb11u1) на замену (2023c-5) …
dpkg: ошибка при обработке архива /var/cache/apt/archives/tzdata_2025b-0+deb11u1_all.deb (--unpack):
 не удалось открыть «/usr/share/zoneinfo/posix/Africa/Abidjan.dpkg-new»: Нет такого файла или каталога
При обработке следующих пакетов произошли ошибки:
 /var/cache/apt/archives/tzdata_2025b-0+deb11u1_all.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)
Чтение списков пакетов… Готово
Построение дерева зависимостей… Готово
Чтение информации о состоянии… Готово
Уже установлен пакет sudo самой новой версии (1.9.13p3-1).
Некоторые пакеты не могут быть установлены. Возможно, то, что вы просите,
неосуществимо, или же вы используете нестабильную версию дистрибутива, где
запрошенные вами пакеты ещё не созданы или были удалены из Incoming.
Следующая информация, возможно, вам поможет:

Следующие пакеты имеют неудовлетворённые зависимости:
 python3-pyrsistent : Зависит: python3 (< 3.10) но 3.11.2-1+b1 должен быть установлен
 python3-setuptools : Зависит: python3-pkg-resources (= 52.0.0-4+deb11u1) но 66.1.1-1 должен быть установлен
 python3-yaml : Зависит: python3 (< 3.10) но 3.11.2-1+b1 должен быть установлен
E: Невозможно исправить ошибки: у вас зафиксированы сломанные пакеты.
useradd: пользователь «tbot» уже существует
Synchronizing state of docker.service with SysV service script with /lib/systemd/systemd-sysv-install.
Executing: /lib/systemd/systemd-sysv-install enable docker
Введите название каталога где будет хранится бот [bot1]: ^Z
[2]+  Остановлен    bash <(wget --cache=off -q -O - https://github.com/ghzserg/zmod_ff5m/raw/refs/heads/1.5/telegram/telegram.sh)
root@orangepizero3:~#

noyhay1, 2025-05-13:
информация на запуск apt list --upgradable -a
```
root@orangepizero3:~# apt list --upgradable -a
Вывод списка… Готово
tzdata/oldstable-security,oldstable-security 2025b-0+deb11u1 all [может быть обновлён с: 2023c-5]
tzdata/oldstable,oldstable 2024a-0+deb11u1 all
tzdata/now 2023c-5 all [установлен, может быть обновлён до: 2025b-0+deb11u1]
tzdata/oldstable-updates,oldstable-updates 2021a-1+deb11u11 all
```

noyhay1, 2025-05-13:
при запуске sudo apt -y upgrade

```
Чтение списков пакетов… Готово
Построение дерева зависимостей… Готово
Чтение информации о состоянии… Готово
Расчёт обновлений… Готово
Следующие пакеты будут обновлены:
  tzdata
Обновлено 1 пакетов, установлено 0 новых пакетов, для удаления отмечено 0 пакетов, и 0 пакетов не обновлено.
Необходимо скачать 0 B/307 kB архивов.
После данной операции объём занятого дискового пространства возрастёт на 941 kB.
Preconfiguring packages ...
(Чтение базы данных … на данный момент установлено 205311 файлов и каталогов.)
Подготовка к распаковке …/tzdata_2025b-0+deb11u1_all.deb …
Распаковывается tzdata (2025b-0+deb11u1) на замену (2023c-5) …
dpkg: ошибка при обработке архива /var/cache/apt/archives/tzdata_2025b-0+deb11u1_all.deb (--unpack):
 не удалось открыть «/usr/share/zoneinfo/posix/Africa/Abidjan.dpkg-new»: Нет такого файла или каталога
При обработке следующих пакетов произошли ошибки:
 /var/cache/apt/archives/tzdata_2025b-0+deb11u1_all.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

noyhay1, 2025-05-13:
Вообщем докладываю, 
на чистой системе debian каталога orangepi

При обработке следующих пакетов произошли ошибки:
 /var/cache/apt/archives/tzdata_2025b-0+deb11u1_all.deb

помолго apt purge tzdata

по 
Следующие пакеты имеют неудовлетворённые зависимости:
 python3-pyrsistent : Зависит: python3 (< 3.10) но 3.11.2-1+b1 должен быть установлен
 python3-setuptools : Зависит: python3-pkg-resources (= 52.0.0-4+deb11u1) но 66.1.1-1 должен быть установлен
 python3-yaml : Зависит: python3 (< 3.10) но 3.11.2-1+b1 должен быть установлен

помогло sudo apt --fix-missing purge $(dpkg -l | grep 'python3\.1[01]' | awk '{print $2}')

и заново запуск bash <(wget --cache=off -q -O - https://github.com/ghzserg/zmod_ff5m/raw/refs/heads/1.5/telegram/telegram.sh)

готовлю документацию с пояснениями

noyhay1, 2025-05-13:
Далее опять ошибки...
```
Введите название каталога где будет хранится бот [bot1]: noyhay
Бот установлен в каталог /home/tbot
1. Идете к https://t.me/BotFather
2. /newbot
3. Вводите любое имя, которое вам нравится
4. Вводите имя бота например ff5msuper_bot - обязательно _bot в конце.
5. Получаете длинный ID - его нужно будет прописать в настройках бота в параметр bot_token
Введите bot_token: 
Creating network "noyhay_default" with the default driver
Pulling moonraker-telegram-bot (lefskiy/moonraker-telegram-bot:latest)...
latest: Pulling from lefskiy/moonraker-telegram-bot
7ce705000c39: Pull complete
d02d1a1ced20: Pull complete
b505734ccb3b: Pull complete
c64f5fd9b3eb: Pull complete
acded0e49130: Pull complete
4f4fb700ef54: Pull complete
7a501a00b6ad: Pull complete
75cdc5216cb6: Pull complete
10e17942c658: Pull complete
Digest: sha256:2ccf6fd8e3324ade124afe1ce85d85c31ccbfe67a4c1d6fdaed0be41ef9131af
Status: Downloaded newer image for lefskiy/moonraker-telegram-bot:latest
Pulling cron (alpine:latest)...
latest: Pulling from library/alpine
6e771e15690e: Pull complete
Digest: sha256:a8560b36e8b8210634f77d9f7f9efd7ffa463e380b75e2e74aff4511df3ef88c
Status: Downloaded newer image for alpine:latest
Creating noyhay_cron_1 ...
Creating noyhay_moonraker-telegram-bot_1 ... error

ERROR: for noyhay_moonraker-telegram-bot_1  Cannot start service moonraker-telegram-bot: AppArmor enabled on system but the docker-default profile could not be loaded: running `apparmor_parser apparmor_parser --version` failed with outpuCreating noyhay_cron_1                   ... error
error: exec: "apparmor_parser": executable file not found in $PATH

ERROR: for noyhay_cron_1  Cannot start service cron: AppArmor enabled on system but the docker-default profile could not be loaded: running `apparmor_parser apparmor_parser --version` failed with output:
error: exec: "apparmor_parser": executable file not found in $PATH

ERROR: for moonraker-telegram-bot  Cannot start service moonraker-telegram-bot: AppArmor enabled on system but the docker-default profile could not be loaded: running `apparmor_parser apparmor_parser --version` failed with output:
error: exec: "apparmor_parser": executable file not found in $PATH

ERROR: for cron  Cannot start service cron: AppArmor enabled on system but the docker-default profile could not be loaded: running `apparmor_parser apparmor_parser --version` failed with output:
error: exec: "apparmor_parser": executable file not found in $PATH
ERROR: Encountered errors while bringing up the project.
Заходите в своего бота, через телеграм
Он напишет. Unauthorized access detected with chat_id:
Впишите полученное числю в chat_id
Введите chat_id:
Removing noyhay_moonraker-telegram-bot_1 ... done
Removing noyhay_cron_1                   ... done
Removing network noyhay_default
Creating network "noyhay_default" with the default driver
Creating noyhay_cron_1 ...
Creating noyhay_moonraker-telegram-bot_1 ... error

ERROR: for noyhay_moonraker-telegram-bot_1  Cannot start service moonraker-telegram-bot: AppArmor enabled on system but the docker-default profile could not be loaded: running `apparmor_parser apparmor_parser --version` failed with outpuCreating noyhay_cron_1                   ... error
error: exec: "apparmor_parser": executable file not found in $PATH

ERROR: for noyhay_cron_1  Cannot start service cron: AppArmor enabled on system but the docker-default profile could not be loaded: running `apparmor_parser apparmor_parser --version` failed with output:
error: exec: "apparmor_parser": executable file not found in $PATH

ERROR: for moonraker-telegram-bot  Cannot start service moonraker-telegram-bot: AppArmor enabled on system but the docker-default profile could not be loaded: running `apparmor_parser apparmor_parser --version` failed with output:
error: exec: "apparmor_parser": executable file not found in $PATH

ERROR: for cron  Cannot start service cron: AppArmor enabled on system but the docker-default profile could not be loaded: running `apparmor_parser apparmor_parser --version` failed with output:
error: exec: "apparmor_parser": executable file not found in $PATH
ERROR: Encountered errors while bringing up the project.
Нужно создать еще одного бота? [y/N]:
```

noyhay1, 2025-05-13:
При ошибках
error: exec: "apparmor_parser": executable file not found in $PATH
Выполнить:
apt install apparmor apparmor-utils

И пообеда))) неужели так сложно должно быть?

ghzserg, 2025-05-14:
Я не оказываю тех поддержку по дебиану. 
У вас определенно некорректно установлен/настроен дебиан.
Если у вас не работает ```apt install docker.io docker-compose docker sudo -y``` то это точно не проблема мода

closed 2025-05-14 by ghzserg