ghzserg/zmod #117: Предложение функционала

Jeinerox, 2025-05-31:
- [X] У меня AD5M
- [ ] У меня AD5M PRO
- [ ] У меня AD5X

- [X] Я работаю с родным экраном
- [ ] Я работаю без экрана
- [ ] Я работаю с Guppyscreen

Каким-то образом используя стоковый алгоритм очистки сопла мой 5m расцарапал стол. Почему так произошло я не понимаю, кампы не включал, а стол измерял только в ручном режиме через экран или через мейнсейл по кнопке в разделе с картой стола.

В общем предложение такое: а что если добавить возможность включать NOZZLE_CONTROL для стокового алгоритма очистки сопла _CLEAR_NOZZLE? Естественно не по дефолту, а как возможность, чтобы исключить вероятность порчи стола.

Можно ли как-то вручную временно добавить эту функцию? Я пытался через vim вызвать ZCONTROL_ON перед вызовом _CLEAR_NOZZLE в CLEAR_NOZZLE, но это не сработало, возможно из-за того, что ff5m.cfg является ссылкой или еще чего.
У меня осталась лишь одна целая сторона стола и теперь очень страшно ее использовать без такого предохранителя

Jeinerox, 2025-05-31:
Как понял что не сработало добавление ZCONTROL_ON:
тыкал в стол во время прочистки, защита не сработала
[image]

Jeinerox, 2025-05-31:
Защита настроена на вес 1500. Если я включу ее вручную и надавлю на стол, то она срабатывает - клиппер выключается, а во флюиде вырубается интерфейс. Во время выполнения подобных команд эта защита работает по другому?

На графике температуры при срабатывании защиты должен выключиться подогрев? 

Про переопределение спасибо, дичайше извиняюсь, не увидел этого в доках, хотя уже несколько дней в них роюсь)

Jeinerox, 2025-05-31:
Попробовал через user.cfg. Судя по логам защита включается, но никакой реакции не было при попытке вызвать срабатывание защиты. Давил я точно больше 1.5кг

ghzserg, 2025-05-31:
Считаете что ошибка и работает не так как нужно. Делайте полноценный тикет.

ghzserg, 2025-06-01:
Установите версию 1.5.2-11-gd3876c93
Там включен контроль при печати файлов с родного экрана принтера

closed 2025-06-02 by ghzserg