ghzserg/zmod #119: Проблема

koltinov, 2025-06-03:
Перед тем, как задавать вопрос прочитайте:
- **[Алгоритм оформления TAR_CONFIG](https://github.com/ghzserg/zmod/wiki/FAQ#help)** - без него тикет рассматриваться не будет
- [FAQ](https://github.com/ghzserg/zmod/wiki/FAQ)
- [Рекомендации](https://github.com/ghzserg/zmod/wiki/Recomendations)
- [Документацию по макросам](https://github.com/ghzserg/zmod/wiki/Macros)

- [ ] У меня AD5M
- [ X ] У меня AD5M PRO
- [ ] У меня AD5X

- [ ] Я работаю с родным экраном
- [ ] Я работаю без экрана
- [ X ] Я работаю с Guppyscreen
(замените пробел на x внутри скобок)

Я загрузил [TAR_CONFIG](https://github.com/ghzserg/zmod/wiki/FAQ#help) в чат zmod (https://t.me/FF_5M_5M_Pro/408369) или на файлообменник.
Прикрепите ссылку на сообщение с файлом или прикрепите файл к этому тексту используя кнопку со скрепкой внизу.

Если проблема проявляется на конкретном файле с gcode, то приложите этот файл тоже. 
Если проблема проявляется даже при печати стандартного кубика, то приложите файл с gcode этого кубика.

Я использую стартовый код:

```
PUBLISH_END PAYLOAD=print
START_PRINT EXTRUDER_TEMP=[nozzle_temperature_initial_layer] BED_TEMP=[bed_temperature_initial_layer_single] FORCE_KAMP=False SKIP_LEVELING=True MESH=ABS04
SET_PRINT_STATS_INFO TOTAL_LAYER=[total_layer_count]
AIR_CIRCULATION_STOP
AIR_CIRCULATION_INTERNAL
SET_SERVO SERVO=my_servo ANGLE=75
```

Я написал завершающий код:

```
PUBLISH_END PAYLOAD=end
SET_SKEW CLEAR=1
END_PRINT
```

Вот мои параметры SAVE_ZMOD_DATA:

```
12:30:16 
info Параметры используемые при начале печати, построении карты стола [START_PRINT]
12:30:16 
info Параметры используемые при начале печати, построении карты стола [START_PRINT]
12:30:16 
// Не строить карту стола средствами родного экрана // SAVE_ZMOD_DATA PRINT_LEVELING=0
12:30:16 
// Не строить карту стола средствами родного экрана // SAVE_ZMOD_DATA PRINT_LEVELING=0
12:30:16 
// Где возможно использовать адаптивную карту стола(KAMP), вместо полной карты стола // SAVE_ZMOD_DATA USE_KAMP=1
12:30:16 
// Тестировать карту стола перед печатью // SAVE_ZMOD_DATA MESH_TEST=1
12:30:16 
// Играть MIDI при запуске печати: my6.mid // SAVE_ZMOD_DATA MIDI_START=my6.mid
12:30:16 
// Играть MIDI при паузе печати: my2.mid // SAVE_ZMOD_DATA MIDI_PAUSE=my2.mid
12:30:16 
// Проверять MD5 сумму файла // SAVE_ZMOD_DATA FORCE_MD5=1
12:30:16 
// SKEW коррекция отключена // SAVE_ZMOD_DATA DISABLE_SKEW=1
12:30:16 
// Загрузка Z-OFFSET из глобальных параметров // SAVE_ZMOD_DATA LOAD_ZOFFSET=1
12:30:16 
// Включить очистку сопла линией // SAVE_ZMOD_DATA DISABLE_PRIMING=0
12:30:16 
// Не использовать предочистку сопла // SAVE_ZMOD_DATA PRECLEAR=0
12:30:16 
// Алгоритм очистки: MY_CLEAR_2 // SAVE_ZMOD_DATA CLEAR=MY_CLEAR_2
12:30:16 
info Параметры используемые при окончании и отмене печати [END_PRINT]
12:30:16 
// Играть MIDI в конце печати: my.mid // SAVE_ZMOD_DATA MIDI_END=my.mid
12:30:16 
// Автоматически закрывать диалоги на родном экране через 20 секунд (быстро) // SAVE_ZMOD_DATA CLOSE_DIALOGS=2
12:30:16 
// Автоматически выключать моторы после печати/отмены печати через 25 секунд // SAVE_ZMOD_DATA STOP_MOTOR=1
12:30:16 
// Не перезапускать принтер автоматически // SAVE_ZMOD_DATA AUTO_REBOOT=0
12:30:16 
// Выключать FF5M Pro через (мин): 30 // SAVE_ZMOD_DATA PRO_POWEROFF_TIMEOUT=30
12:30:16 
info Общесистемные параметры
12:30:16 
// Сохранять данные для восстановления печати // SAVE_ZMOD_DATA SAVE_RESTORE=1
12:30:16 
// Используется датчик наличия филамента // SAVE_ZMOD_DATA MOTION_SENSOR=0
12:30:16 
// Использовать SCV из конфигурации при построении графиков шейперов // SAVE_ZMOD_DATA FIX_SCV=1
12:30:16 
// Исправить ошибку E0011 // SAVE_ZMOD_DATA FIX_E0011=1
12:30:17 
// Исправить ошибку E0017 // SAVE_ZMOD_DATA FIX_E0017=1
12:30:17 
// Яркость LED при включении: 100 // SAVE_ZMOD_DATA LED=100
12:30:17 
// Играть MIDI при включении: my5.mid // SAVE_ZMOD_DATA MIDI_ON=my5.mid
12:30:17 
// Использовать стандартный SAVE_CONFIG // SAVE_ZMOD_DATA NEW_SAVE_CONFIG=0
12:30:17 
// SWAP используется на EMMC // SAVE_ZMOD_DATA USE_SWAP=1
12:30:17 
// Запретить китайские облака // SAVE_ZMOD_DATA CHINA_CLOUD=0
12:30:17 
// Сохранять изменения кнопок Fluidd/Mainsail // SAVE_ZMOD_DATA SAVE_MOONRAKER=1
12:30:17 
// Выключать родной экран через (сек): 10 // SAVE_ZMOD_DATA DISPLAY_OFF_TIMEOUT=10
12:30:17 
// Используется родной Klipper
12:30:17 
// Приоритет процесса Klipper (1-минимальный, 40-максимальный): 20
12:30:18 
// Вес: 1250; Контроль настроен и не активен.
12:30:18 
// При сработке отключается Klipper. // ZCONTROL_ABORT
12:30:18 
// Доступные языки: en, ru, de, fr, it, es, ja, ko, zh // LANG LANG=ru
12:30:18 
// zversion:
12:30:18 
// Установлено с флешки: 1.5.0. Обновление с Fluidd/Mainsaill: 1.5.2

​

```

странно работает пауза простая и для смены прутка. для воспроизведения проблемы поставил паузу смены прутка перед вторым слоем. после первого слоя встал на паузу как положено, после нажатия "возобновить" подъехал к детали, коснулся соплом и опять поехал на паузу. нажал еще раз "возобновить", опять подъехал, коснулся, поехал в угол на рабочей подаче без опускания стола, из угла опять к детали и продолжил печать на 25%, но я прождал 6 минут - скорость так и не восстановилась, отменил печать.

до этого ставил паузу по М25, там еще хуже, после 2 раз паузы при печати остывает экструдер.

[config.zip](https://github.com/user-attachments/files/20564179/config.zip)

ghzserg, 2025-06-03:
Судя по логу у вас используется макрос `AFTER_LAYER_CHANGE` - которого нет в моде.
По поводу  паузы. 
1. Принтер сохраняет значение температуры стола, температуры экструдера, скорости
2. Отключается экструдер
3. Отключается контроль сопла
4. Через 10 секунд паркуется голова в угол

При восстановлении:
1. Восстанавливается температура стола и экструдера
2. Восстанавливается скорость 25% от исходной
3. Через 60 секунд восстанавливается скорость 50% от исходной
4. Через 60 секунд восстанавливается скорость 100% от исходной

Сделал апдейт который выводит скорость - обновитесь

koltinov, 2025-06-03:
попробую обновиться и сделать проверку системы

ghzserg, 2025-06-03:
> 
> это же вроде стандартный комментарий в орке? он и в базовом профиле принтера такой же
> 

Именно что комментарий. Но в логах четко видно ```[07:50:10.033] // Unknown command:"AFTER_LAYER_CHANGE"```

koltinov, 2025-06-03:
обновил, сделал проверку, несколько ссылок было неверных, запустил еще раз, пауза, после сразу после начала движения из угла играет миди второй паузы, сопло доезжает до детали и обратно в угол, снимаю с паузы, касается детали соплом и не опуская стол едет в угол, потом опять к детали, и продолжает печать на скорости 6,5, потом 13, потом остается на 25%.

koltinov, 2025-06-03:
`16:25:34 
$ RESUME
16:25:34 
echo: Восстанавливаю "extruder" температуру до 275.0/100.0°C, this may take some time
16:25:58 
// idle_timeout: Timeout set to 300.00 s
16:25:58 
// ZCONTROL_ON. 1250.0. ABORT
16:26:00 
info Speed0: 25.0
16:26:00 
// Unknown command:"STL_ID_0"
16:26:00 
// ZCONTROL_OFF
16:26:00 
// idle_timeout: Timeout set to 86400.00 s
16:26:00 
// audio_midi:
16:26:18 
$ RESUME
16:26:19 
echo: Восстанавливаю "extruder" температуру до 275.0/100.0°C, this may take some time
16:26:39 
// idle_timeout: Timeout set to 300.00 s
16:26:39 
// ZCONTROL_ON. 1250.0. ABORT
16:26:40 
info Speed0: 6.25
16:26:40 
info Отвожу голову в дальний правый угол
16:27:43 
info Speed1: 12.5
16:28:41 
info Speed2: 25`

koltinov, 2025-06-03:
// Unknown command:"STL_ID_0"
таких у меня тоже нет макросов))

koltinov, 2025-06-03:
[config.zip](https://github.com/user-attachments/files/20572786/config.zip)

ghzserg, 2025-06-03:
Ошибка в Klipper. Обновитесь до 1.5.3 и ваш старый файл должен печататься корректно

koltinov, 2025-06-03:
да теперь всё работает корректно. Спасибо!

closed 2025-06-03 by koltinov