ghzserg/zmod #133: Enable Bash on AD5X zmod

f00d4tehg0dz, 2025-07-17:
- [ ] I have AD5M
- [ ] I have AD5M PRO
- [x] I have AD5X

- [X] I work with the native screen
- [ ] I work without a screen
- [ ] I work with Guppyscreen

To install and configure [Happy Hare](https://github.com/moggieuk/Happy-Hare/wiki/Installation) I would need /bin/bash to run the script. Is it possible to add bash support? I see its available in the 5M zmod. 

```
cd /usr/data/Happy-Hare/
./install.sh -i
```

Thanks!

f00d4tehg0dz, 2025-07-18:
Ah ok. Yeah unfortunately that won't work. Darn. 

Previously with the AD5M Pro and xblax mod I had luck modifying the build root and using the ERCF with Happy Hare. Was hoping to give the AD5X a try with an ERCF :)

ghzserg, 2025-07-19:
I am on vacation, without access to a computer. Specify which packages are needed and I will add them after returning from vacation

ghzserg, 2025-09-06:
Bash is enabled in version 1.6.1. GCC cannot be enabled for the mips processor