ghzserg/zmod #136: Проблема GuppyScreen умирает при открытии Spoolman

nazbav, 2025-07-30:
- [ X] У меня AD5M
- [X ] Я работаю с Guppyscreen

https://t.me/FF_5M_5M_Pro/408369/906665
https://drive.google.com/drive/folders/18IHnrhIHcwZlK3PuoUepWohPWUxVyvMf?usp=sharing

Подробное описание ошибки:
При клике на Spoolman, гуппи повисает. Spoolman работает во fluid при этом нормально (живет он, спулман, на внешнем хосте в локальной сети). По логике должен быть список катушек (либо не быть кнопки)
config.tar пока отсутствует, сливает токены от smarthings (розетка на выключение питания принтера).

Вот мои параметры SAVE_ZMOD_DATA:
```
:50:54  info Параметры zmod. Задаются через SAVE_ZMOD_DATA
01:50:54  info Вы работаете БЕЗ родного экрана c GuppyScreen // DISPLAY_OFF GUPPY=1
01:50:54  info Параметры используемые при начале печати, построении карты стола [START_PRINT]
01:50:54  // При каждой печати строить карту стола. // SAVE_ZMOD_DATA PRINT_LEVELING=1
01:50:54  // Где возможно использовать адаптивную карту стола(KAMP), вместо полной карты стола // SAVE_ZMOD_DATA USE_KAMP=1
01:50:54  // Тестировать карту стола перед печатью // SAVE_ZMOD_DATA MESH_TEST=1
01:50:54  // Играть MIDI при запуске печати:  // SAVE_ZMOD_DATA MIDI_START=
01:50:54  // Играть MIDI при паузе печати:  // SAVE_ZMOD_DATA MIDI_PAUSE=
01:50:54  // Проверять MD5 сумму файла // SAVE_ZMOD_DATA FORCE_MD5=1
01:50:54  // SKEW коррекция включена. Профиль skew_profile // SAVE_ZMOD_DATA DISABLE_SKEW=0
01:50:54  // Загрузка Z-OFFSET из глобальных параметров // SAVE_ZMOD_DATA LOAD_ZOFFSET=1
01:50:54  // Включить очистку сопла линией // SAVE_ZMOD_DATA DISABLE_PRIMING=0
01:50:54  // Не использовать предочистку сопла // SAVE_ZMOD_DATA PRECLEAR=0
01:50:54  // Алгоритм очистки: LINE_PURGE // SAVE_ZMOD_DATA CLEAR=LINE_PURGE
01:50:54  info Параметры используемые при окончании и отмене печати [END_PRINT]
01:50:54  // Играть MIDI в конце печати:  // SAVE_ZMOD_DATA MIDI_END=
01:50:54  // Автоматически закрывать диалоги на родном экране через 20 секунд (быстро) // SAVE_ZMOD_DATA CLOSE_DIALOGS=2
01:50:54  // Автоматически выключать моторы после печати/отмены печати через 25 секунд // SAVE_ZMOD_DATA STOP_MOTOR=1
01:50:55  // Не перезапускать принтер автоматически // SAVE_ZMOD_DATA AUTO_REBOOT=0
01:50:55  // Выключать FF5M Pro через (мин): 0 // SAVE_ZMOD_DATA PRO_POWEROFF_TIMEOUT=0
01:50:55  info Общесистемные параметры
01:50:55  // Сохранять данные для восстановления печати // SAVE_ZMOD_DATA SAVE_RESTORE=1
01:50:55  // Используется датчик движения филамента // SAVE_ZMOD_DATA MOTION_SENSOR=1
01:50:55  // Использовать SCV из конфигурации при построении графиков шейперов // SAVE_ZMOD_DATA FIX_SCV=1
01:50:55  // Исправить ошибку E0011 // SAVE_ZMOD_DATA FIX_E0011=1
01:50:55  // Исправить ошибку E0017 // SAVE_ZMOD_DATA FIX_E0017=1
01:50:55  // Яркость LED при включении: 100 // SAVE_ZMOD_DATA LED=100
01:50:55  // Играть MIDI при включении:  // SAVE_ZMOD_DATA MIDI_ON=
01:50:55  // Использовать стандартный SAVE_CONFIG // SAVE_ZMOD_DATA NEW_SAVE_CONFIG=0
01:50:55  // SWAP используется на EMMC // SAVE_ZMOD_DATA USE_SWAP=1
01:50:55  // Запретить китайские облака // SAVE_ZMOD_DATA CHINA_CLOUD=0
01:50:55  // Использовать настройки кнопок Fluidd/Mainsail из ZMOD // SAVE_ZMOD_DATA SAVE_MOONRAKER=0
01:50:55  // Выключать родной экран через (сек): 10 // SAVE_ZMOD_DATA DISPLAY_OFF_TIMEOUT=10
01:50:55  // Используется родной Klipper
01:50:55  // Приоритет процесса Klipper (1-минимальный, 40-максимальный): 20
01:50:55  // Вес: 1500; Контроль настроен и не активен.
01:50:56  // При сработке отключается Klipper. // ZCONTROL_ABORT
01:50:56  // Доступные языки: en, ru, de, fr, it, es, ja, ko, zh // LANG LANG=ru
01:50:56  // zversion:
```

ghzserg, 2025-08-01:
Не хватает tar_config

ghzserg, 2025-08-01:
[image]

ghzserg, 2025-08-01:
https://github.com/ghzserg/zmod/wiki/Telegram_ru#spoolman

ghzserg, 2025-08-01:
У меня нет возможности искать какая из катушек у вас в бд выдает сбой. Если найдете и сделаете компактную базу, только со ьойной катушкой - посмотрю и поправлю. Возможно просто оперативной памяти не хватает.

mortonlame, 2025-08-03:
Ты б скинул тар конфиг. да, повисает наглухо. но при этом, 20 катушек моих - нормально так скроллятся без зависаний
тоже подозреваю нехватку оперативной памяти.

mortonlame, 2025-08-03:
в базе у тебя есть упоминания импортированных катушек из твоего калькулятора. а не может быть там каких-то кривых данных?

nazbav, 2025-08-03:
> Ты б скинул тар конфиг. да, повисает наглухо. но при этом, 20 катушек моих - нормально так скроллятся без зависаний
> тоже подозреваю нехватку оперативной памяти. 

zmod.zip прикреплен давно в тексте — это tar, github не дает загрузить тар, ток зипы.