ghzserg/zmod #138: датчик движения филамента,

noyhay1, 2025-08-03:
[ ] У меня AD5M
[ ] Я работаю с Guppyscreen
https://cloud.mail.ru/public/9qz1/MtdTi26Fh конфиг

При вызове с параметром 1 используется датчик движения филамента?
При этом параметре печатает всегда и не находит датчик.
На родном экране отключил датчик
Делал вот этот датчик https://www.thingiverse.com/thing:6841711

**Помогло редактирование**
[filament_switch_sensor e1_sensor]
pause_on_runout: False
switch_pin: PE1
event_delay: 1.0

на

[filament_switch_sensor e1_sensor]
pause_on_runout: False
switch_pin: !PE1
event_delay: 1.0

ghzserg, 2025-08-04:
У вас используется датчик наличия филамента // SAVE_ZMOD_DATA MOTION_SENSOR=0

А не датчик движения филамента.
Также ваше основное изменение, это полярность выхода

noyhay1, 2025-08-04:
`У вас используется датчик наличия филамента // SAVE_ZMOD_DATA MOTION_SENSOR=0`

Лог старый, вы правы, работает с MOTION_SENSOR=1 который и используется.
Насчёт полярности. Можете добавить это в примечание, обычному человеку это пригодится.

Подключал по схеме