ghzserg/zmod #141: Problem:Installing octoapp companion plugins.

cackmobile, 2025-08-20:
- [ ] I have AD5M
- [ ] I have AD5M PRO
- [X] I have AD5X

- [X] I work with the native screen
- [ ] I work without a screen
- [ ] I work with Guppyscreen
(replace the space with x inside the brackets)

I wantto use octoapp or the moonraker one but I need to install teh companion plugins. I've tried all methods via SSH but nothing seems to work. Can someone please help

[config.zip](https://github.com/user-attachments/files/21892001/config.zip)

`

cackmobile, 2025-08-20:
Hi le

cackmobile, 2025-08-21:
> I tried it and it's asks for the companion plugins
>
> Regards
>
> Elliott

Oh wow, I think adding the port solved the issue!

Regards

Elliott

cackmobile, 2025-08-24:
HI again, I also worked out you can install octoapp and mobileraker on a network connected device like a Pi. which i've done and is now working

ghzserg, 2025-08-25:
It's great that everything works on the external server.