ghzserg/zmod #158: Problem Changing filaments with IFS

not-crazyramen, 2025-09-13:
- [ ] I have AD5M
- [ ] I have AD5M PRO
- [x] I have AD5X

- [ ] I work with the native screen
- [ ] I work without a screen
- [x] I work with Guppyscreen

TAR_CONFIG LINK: https://drive.google.com/drive/folders/1vltMElR8XqL4HN_jANdx232KaJfJhg1a?usp=sharing

I use the start code:

```
Put your printer start G-code here
START_PRINT EXTRUDER_TEMP=[nozzle_temperature_initial_layer] BED_TEMP=[bed_temperature_initial_layer_single]
SET_PRINT_STATS_INFO TOTAL_LAYER=[total_layer_count]

```

I wrote the end code:

```
END_PRINT
```

Here are my SAVE_ZMOD_DATA parameters:

```
 11:26:41  // Turn off native screen after (sec): 30 // SAVE_ZMOD_DATA DISPLAY_OFF_TIMEOUT=30
11:26:40  // Use Fluidd/Mainsail button settings from ZMOD // SAVE_ZMOD_DATA SAVE_MOONRAKER=0
11:26:40  // Allow Chinese clouds // SAVE_ZMOD_DATA CHINA_CLOUD=1
11:26:40  // Play MIDI at power on:  // SAVE_ZMOD_DATA MIDI_ON=
11:26:40  // LED brightness at power on: 0 // SAVE_ZMOD_DATA LED=0
11:26:40  // Fix E0017 error // SAVE_ZMOD_DATA FIX_E0017=1
11:26:40  // Don't fix E0011 error // SAVE_ZMOD_DATA FIX_E0011=0
11:26:40  // Use SCV from config for shaper graphs // SAVE_ZMOD_DATA FIX_SCV=1
11:26:40  // Insert rod automatically // SAVE_ZMOD_DATA AUTOINSERT=1
11:26:40  // Don't hide color selection when printing starts // SAVE_ZMOD_DATA SILENT=0
11:26:40  // Save data for print recovery // SAVE_ZMOD_DATA SAVE_RESTORE=1
11:26:40  info System-wide parameters
11:26:40  // Don't reboot printer automatically // SAVE_ZMOD_DATA AUTO_REBOOT=0
11:26:40  // Auto disable motors after print/cancel in 25 sec // SAVE_ZMOD_DATA STOP_MOTOR=1
11:26:40  // Don't close dialogs on native screen // SAVE_ZMOD_DATA CLOSE_DIALOGS=0
11:26:40  // Play MIDI at print end:  // SAVE_ZMOD_DATA MIDI_END=
11:26:40  info Parameters used at print end and cancellation [END_PRINT]
11:26:40  // Purge algorithm: LINE_PURGE // SAVE_ZMOD_DATA CLEAR=LINE_PURGE
11:26:40  // Enable nozzle line purge // SAVE_ZMOD_DATA DISABLE_PRIMING=0
11:26:40  // Don't load Z-OFFSET from global settings // SAVE_ZMOD_DATA LOAD_ZOFFSET=0
11:26:40  // SKEW correction disabled // SAVE_ZMOD_DATA DISABLE_SKEW=1
11:26:40  // Check file MD5 checksum // SAVE_ZMOD_DATA FORCE_MD5=1
11:26:40  // Play MIDI on print pause:  // SAVE_ZMOD_DATA MIDI_PAUSE=
11:26:40  // Play MIDI on print start: halo_respawn.mid // SAVE_ZMOD_DATA MIDI_START=halo_respawn.mid
11:26:40  // Test bed mesh before printing // SAVE_ZMOD_DATA MESH_TEST=1
11:26:40  // Use adaptive mesh (KAMP) where possible instead of full bed mesh // SAVE_ZMOD_DATA USE_KAMP=1
11:26:40  // Build bed mesh on each print. // SAVE_ZMOD_DATA PRINT_LEVELING=1
11:26:40  info Parameters used at print start and bed leveling [START_PRINT]
11:26:40  info You are working WITHOUT native screen with GuppyScreen // DISPLAY_OFF GUPPY=1
11:26:40  info ZMOD parameters. Set via SAVE_ZMOD_DATA
```

I'm trying to print a gcode file with a single color. The color I want to use is not the currently loaded filament. When the printer changes the filament with the IFS to the correct one, I get errors. This only happens with DISPLAY_OFF. This occurs on unloading. When I use the COLOR macro to unload filament this happens also. If I use the native display to manually do it, it unloads and loads correctly. 

These are the errors in the console. I have attached my printer.log file at the link with the config.tar

```11:25:43  !! Internal error on command:"PRINT_ZCOLOR"
11:25:43  !! Internal error on command:"SET_CURRENT_PRUTOK"
11:25:43  !! Internal error on command:"_A_CHANGE_FILAMENT"
11:25:43  !! Internal error on command:"_CHANGE_FILAMENT"
11:25:43  !! Ошибка при смене филамента: name 'need_stop' is not defined
Печать отменена
11:25:43  !! Internal error on command:"INSERT_PRUTOK_IFS"
11:25:43  !! Internal error on command:"_INSERT_PRUTOK_IFS"
11:25:43  !! Internal error on command:"IFS_REMOVE_CURRENT_PRUTOK"
11:25:43  !! Internal error on command:"IFS_REMOVE_PRUTOK"
```

ghzserg, 2025-09-14:
I saw the following problems with you
1. No auto map
https://github.com/ghzserg/zmod/wiki/FAQ_en#what-options-are-available-for-bed-leveling
2. There are homing errors
https://github.com/ghzserg/zmod/wiki/Global_en#fix_e0011
3. Update the zmod version to `**1.6.1-53-gc6784c21**`

ninjamida, 2025-09-14:
I'm also having the same issues. I double-checked that auto map exists, and enabled the E0011 fix, and installed the 1.6.1.53 update, but I still get this issue when trying to unload filament (or switch during a print) while native screen is disabled.

Will get logs etc later, but for now - I can also note a couple of other things that may or may not be relevant here:
1. I periodically get an error "19:17:34 // IFS: sensor error: read failed: device reports readiness to read but returned no data (device disconnected or multiple access on port?)". This mostly happens after the printer starts up.
2. If I run DISPLAY_OFF GUPPY=1 and then restart the printer, it has the native screen rather than GuppyScreen. This does not go away if I wait a while (tried about 5 minutes). If I then run DISPLAY_OFF GUPPY=1 again, GuppyScreen works as normal.

ghzserg, 2025-09-14:
The error "19:17:34 // IFS: sensor error: read failed: the device reports being ready to read, but has not returned data (is the device disconnected or is there multiple access to the port?)" occurs periodically. Most often, this happens after starting the printer.

This is normal. Because the native screen starts at the first stage. And via https://github.com/ghzserg/zmod/wiki/Global_ru#display_off_timeout
Guppy
Specify SAVE_ZMOD_DATA DISPLAY_OFF_TIMEOUT=20

ninjamida, 2025-09-14:
I have tested this update.

It gets past the point where it previously crashed. When the nozzle reaches 250 degrees, the printer cuts the filament, then moves the print head back to the waste chute. At this point, I immediately get an error message in Russian: "Не удалось извлечь пруток из экструдера". There does not appear to be any useful information in printer.log - just this error message (four times) inbetween Stats lines.

ghzserg, 2025-09-14:
Need TAR_CONFIG

ghzserg, 2025-09-14:
And check what the filament presence sensor in the extruder shows. If there is filament left in the extruder, then this message appears and this is normal. You need to clean the extruder.

[image]

[image]

[image]

ghzserg, 2025-09-14:
I need a tar. Because the verbal description does not give what is needed. When cutting the rod, it should reach the basket and start extracting the rod with an extruder and IFS. If this does not happen, then you need to look at the tar.

ghzserg, 2025-09-14:
Insert into `mod_data/user.cfg`
```
[zmod_ifs]
debug: True
```

ninjamida, 2025-09-14:
```
 22:22:32  $ COLOR
22:22:32  // action:prompt_end
22:22:32  // action:prompt_begin Loaded material
22:22:32  // action:prompt_text Extruder: 1: PETG/black
22:22:32  // action:prompt_text IFS: True
22:22:32  // action:prompt_text Select a spool to modify
22:22:32  // action:prompt_button_group_start
22:22:32  // action:prompt_button 1: PETG/black|RUN_ZCOLOR SLOT=1 HEX=161616 TYPE=PETG|primary
22:22:32  // action:prompt_button 2: PETG/white|RUN_ZCOLOR SLOT=2 HEX=FFFFFF TYPE=PETG|primary
22:22:32  // action:prompt_button_group_end
22:22:32  // action:prompt_footer_button Ok|RESPOND TYPE=command MSG=action:prompt_end
22:22:32  // action:prompt_footer_button Reset colors|RESET_ZCOLOR
22:22:32  // action:prompt_show
22:22:34  $ RUN_ZCOLOR SLOT=2 HEX=FFFFFF TYPE=PETG
22:22:34  // action:prompt_end
22:22:34  // action:prompt_begin Select action
22:22:34  // action:prompt_text Spool 2: PETG/white
22:22:34  // action:prompt_button_group_start
22:22:34  // action:prompt_button Change color|CHANGE_ZCOLOR SLOT=2 TYPE=PETG|primary
22:22:34  // action:prompt_button Change type|CHANGE_ZCOLOR SLOT=2 HEX=FFFFFF|primary
22:22:34  // action:prompt_button Load|IN_ZCOLOR SLOT=2 NAPR=0|primary
22:22:34  // action:prompt_button Unload|IN_ZCOLOR SLOT=2 NAPR=1|primary
22:22:34  // action:prompt_button_group_end
22:22:34  // action:prompt_footer_button Cancel|RESPOND TYPE=command MSG=action:prompt_end
22:22:34  // action:prompt_show
22:22:35  $ IN_ZCOLOR SLOT=2 NAPR=0
22:22:35  // action:prompt_end
22:22:35  info Loading filament 2 PETG
22:22:35  info Go to trash
22:22:35  info Extracting filament from the extruder 1 PETG
22:22:35  info Cutting the filament
22:22:40  info Go to trash
22:22:46  // Извлечь пруток 1 длинной 60 со скоростью 1200
22:22:47  // F11 C1 L60 S1200 > F11 ok. FFS channel 1 exiting.
22:22:48  // F13 need:None|5 cur:15 > F13 ok. FFS_state: 15 silk_state: 3 chan: 1 ffs_channels_insert: 0 stall_state: 0 jinsi_GCONF: 000001dc qiehuan_GCONF: 000001dc
22:22:49  // F13 need:None|5 cur:15 > F13 ok. FFS_state: 15 silk_state: 3 chan: 1 ffs_channels_insert: 0 stall_state: 0 jinsi_GCONF: 000001dc qiehuan_GCONF: 000001dc
22:22:50  // F13 need:None|5 cur:5 > F13 ok. FFS_state: 5 silk_state: 3 chan: 1 ffs_channels_insert: 0 stall_state: 0 jinsi_GCONF: 000001dc qiehuan_GCONF: 000001dc
22:22:53  !! Не удалось извлечь пруток из экструдера
22:22:53  !! Не удалось извлечь пруток из экструдера
22:22:53  !! Не удалось извлечь пруток из экструдера
22:22:53  !! Не удалось извлечь пруток из экструдера
22:22:53  !! Не удалось извлечь пруток из экструдера
```

This is when trying to switch the loaded color via the COLOR macro.

Config.tar: https://1drv.ms/u/c/d773beb8dc4d8d6e/EQYI9UN496ZIi1pK8zWs82MBKaqSzXaojSWtwpYweRTbjA?e=LLgDiW

ghzserg, 2025-09-14:
The log clearly shows that the extraction of rod 1 is working using the IFS. Do you really have rod number 1 in your extruder?
If you have a different rod in your extruder, then it is clear why it cannot extract it

ninjamida, 2025-09-14:
I did some more testing of this. First, I made sure no rods at all were in the extruder. Then, I loaded Channel 2 (PETG White). This worked properly.

Then, I tried to load Channel 1 (PETG Black). The printer heated up, cut the filament, returned to the print bucket, then _seemingly_ immediately produced the error message.

However - by paying close attention, I think I figured out what's happening here. The extruder does indeed unload the rod as far as it can - but this is not far enough to un-trip the sensor. What should happen, **I think** (ie: I might be wrong), is that once the extruder motor finishes, the IFS should take over _without_ checking the sensor first.

In particular, one very important detail I noticed: If I load filament (whether native screen or Guppy), then try to pull it out by hand, it's very hard to do because of the extruder gears gripping it. However, if I try to do this after I get the error message, then I can pull it easily. This is why I think the extruder has unloaded the filament as far as it can, but the IFS has failed to take over for the rest of the unload.

A couple of side points here:
1. When the extruder motor unloads the filament, it sounds like it's going at a very fast speed compared to the native screen.
2. When the print head moves from the cutter back to the waste chute, it moves quite slow compared to the native screen.
3. When loading, the printer wipes the nozzle between poops. This doesn't happen on the native screen (but I think it's a good idea!).
4. If possible, when this error happens, maybe it's a good idea to turn the extruder heater off? Currently it remains on.

Updated config.tar just in case: https://1drv.ms/f/c/d773beb8dc4d8d6e/Em6NTdy4vnMggNebAwAAAAABb3z6g2l9_s67hN_lnLktpw?e=HRuhqw

ghzserg, 2025-09-14:
Update to 1.6.1-60/ Translated ,message

ghzserg, 2025-09-14:
Need a tarkonyig with a detailed description of what and where was uploaded. And also the file that you print

not-crazyramen, 2025-09-14:
Alright it does switch rods correctly mid print, however it rammed my print off the bed when it switched. Looks like it raised the bed while switching filaments. This is the first multicolor print I have done. I'm not sure if this was an issue caused by setting my 

New config.tar, gcode file, user.cfg, and printer.log: https://drive.google.com/drive/u/0/folders/1TW6vjju9JjuZD7tPYCp7U3QYiZMZA3nn

I started this print directly from orcaslicer (upload and print). I wanted to print it with PETG

At the start of the print this is the state of the printer:
- rods in IFS: 
    - 2: PETG
    - 3: PLA
- currently loaded rod:
    - 2: PETG

What happened when the print started:
- PRINT menu activates, showing the following:
    - T0 -> 2: PETG
    - T1 -> 3: PLA
    - T2 -> 3: PLA
    - T3 -> 3: PLA
    
- Assign the following filaments:
    - T0 -> 3: PLA
    - T1 -> 3: PLA
    - T2 -> 3: PLA
    - T3 -> 3: PLA
- Print Begins
- Print happens with currently loaded rod - 3: PLA (load cell tare, bed mesh, purge line, etc.)

What I wanted to happen when the print started:
- PRINT menu activates
- Assign the following filaments:
    - T0 -> 3: PLA
    - T1 -> 3: PLA
    - T2 -> 3: PLA
    - T3 -> 3: PLA
- Print Begins
- **ZMOD detects that 2: PETG is loaded and 3: PLA is the filament to be printed**
- **ZMOD unloads 2: PETG and loads 3: PLA**
- Print begins

ninjamida, 2025-09-14:
I don’t have any new info as I just woke up; but I’m wondering - is there any way I can modify the scripts myself without completely reinstalling from USB? (Maybe by SSHing in as root?) I’d be quite happy to try and do a bit of experimenting myself sometime; and I actually have the AD5X and IFS so I can test it as I go.

not-crazyramen, 2025-09-14:
Testing it again and it works. Correctly selected the filament before starting.

I can only write off it not working earlier today as user error - sorry about that :/

not-crazyramen, 2025-09-15:
I'm sure there's a slicker way to do it but I have it fixed by adding

```   G91
   G1 Z-5 F3000
   G90
```
to the beginning 

and 

```
   G91
   G1 Z-5 F3000
   G90
```
to the end of the _A_CHANGE_FILAMENT macro

ninjamida, 2025-09-15:
> I'm sure there's a slicker way to do it but I have it fixed by adding
> 
> ```
>    G1 Z-5 F3000
>    G90
> ```
> 
> to the beginning
> 
> and
> 
> ```
>    G91
>    G1 Z-5 F3000
>    G90
> ```
> 
> to the end of the _A_CHANGE_FILAMENT macro

I suspect you copy/pasted this wrong - the start one should be Z5 (not Z-5) and should have a G91 before it, I think?

Anyway, going to try this out and see what I think of it.

ninjamida, 2025-09-15:
> I'm sure there's a slicker way to do it but I have it fixed by adding
> 
> ```
>    G1 Z-5 F3000
>    G90
> ```
> 
> to the beginning
> 
> and
> 
> ```
>    G91
>    G1 Z-5 F3000
>    G90
> ```
> 
> to the end of the _A_CHANGE_FILAMENT macro

Okay, yeah, once the fixes as mentioned in my last comment are made, this is perfect. I don't know if it's the same exact length as native screen, but it definitely achieves the same result.

Anyway, yeah, as ghzserg says - the original problem for this issue is solved now, so probably time to close this one?

ghzserg, 2025-09-15:
Explain in more detail what problem you want to solve.
You want to lower the table a little before changing the filament - right?
If it's not too much trouble, print one cube with the original screen and one filament change and send me the log file.
I'll see how it works

ghzserg, 2025-09-15:
В общем я сделал обновление. Тестируйте

ninjamida, 2025-09-15:
Seems good to me!

ghzserg, 2025-09-15:
Thank you for finding the error
Error loading template 'gcode_macro _A_CHANGE_FILAMENT:gcode': jinja2.exceptions.TemplateSyntaxError: unexpected char '`' at 608
I fixed it. Please check if everything works as it should
Yes, after a pause everything should work correctly.
But it's better to test everything

not-crazyramen, 2025-09-16:
Awesome work y'all. I just confirmed it's all working as expected. 

Thanks for the link to that other issue too btw it's helped me understands more about how the code is set up

closed 2025-09-16 by not-crazyramen