ghzserg/zmod #164: Suggestion (AD5X, GuppyScreen / headless): Seperate drop lengths for during a print vs outside a print

ninjamida, 2025-09-14:
I love the feature to customize the drop length when changing color / loading filament. However, it would be nice to be able to set different values for the drop length when changing outside of a print (ie: loading filament via COLOR macro, or loading the first color at the start of a print) versus changing during a print (ie: multicolor prints).

The reason for this is: When changing color outside of a print, the waste has nowhere else to go. It just has to go down the waste chute. However, during a print, there's another option: I could disable the drop altogether, and purge solely into wipe tower / infill / supports.

If it's not possible to have seperate lengths, but it is possible to have a "disable drop during print" option, that would also be sufficient.

If you can give me some starters on where to look (ie: what are the relevant functions in the script? What should I look at to detect if it's mid-print or not?) I'm happy to try and code this myself.

ghzserg, 2025-09-15:
The easiest way to do this is through macros. Look at the printer status and if it is printing, skip the reset.

ghzserg, 2025-09-15:
https://github.com/ghzserg/zmod_ff5x/commit/8c7b3309465c5f4a7247fb5ed208329d9aedd2c9

ninjamida, 2025-09-15:
> https://github.com/ghzserg/zmod/wiki/Global_en#use_trash_on_print

Thanks! Going to test this now.

ninjamida, 2025-09-15:
Doesn't work. When the filament is changed, after a few seconds, it reports a filament jam.

There is no actual jam - what's actually happened, it seems, is that the new color hasn't actually loaded into the extruder. I can confirm it's not the old one blocking it - even if I manually pull the old one back further so it can't possibly be in the way, this still happens.

I'm going to experiment a bit with the macro and I'll report back if I figure out something that works.

ninjamida, 2025-09-15:
First feedback: I tried changing the macro so that if use_trash_on_print is 0 (and it's printing), it will execute the following one line:

`_SBROS_TRASH_DAVIM PRUTOK={prutok} FILAMENT_FAN_SPEED={filament_fan_speed} FILAMENT_DROP_LENGTH=10 FILAMENT_DROP_LENGTH_ADD=0 FILAMENT_LOAD_SPEED={filament_load_speed} FILAMENT_UNLOAD_AFTER_DROP=0`

With this change, I no longer get errors, and no actual pooping occurs. The printer doesn't actually poop.

However, when it resumes printing, initially nothing prints - this is because there's a gap between the rod loaded into the extruder, and the old filament that was cut, due to the retraction (as specified in filament.json). So my next test: I'm instead going to set it to drop a length equal to the old filament's total unloading (25mm in my case). I couldn't quite figure out how to actually read this data from the configuration in GCode (I can see how I'd do it in a Python script, but not sure about GCode), so I just hardcoded it for testing purposes - will report back soon.

ninjamida, 2025-09-15:
Yeah, I think the reduction in length is what's needed here though - turning it off altogether, at least based on your original code, just causes it to not load the filament into the extruder.

When the filament reaches the extruder gear, the IFS is not strong enough to push it through - the extruder needs to do the work here. (Even by hand, it's very hard to push it through the extruder, but not impossible.) Likewise for the IFS pulling it out - it can't pull it out of the extruder either.

I have encountered another issue - sometimes (not always), when a filament change occurs, the old filament is not unloaded far enough for the new filament to be inserted - it seems the IFS doesn't unload it any further after the extruder is finished, or if it does, it doesn't unload it very far. Is there a particular setting in filament.json I should change to alter the distance the IFS unloads it? I'm not sure exactly what the distance is on native screen, but I would guess about 40mm to 50mm.

ghzserg, 2025-09-15:
Тогда  я делаю просто - отключаю второй сброс если стоит параментр

ninjamida, 2025-09-15:
I've seen that page, and referred to it already to tweak my settings. However, I don't see a setting specifically for how far the IFS should unload the filament after the extruder is finished.

Do I just increase `filament_unload_after_cutting`, and the IFS will automatically take over once the extruder can't move it anymore?

Or am I misunderstanding `filament_autoinsert_ret_length`? I get the impression that's the length that, when loading a new filament into the IFS, when the IFS has loaded it all the way to the extruder, it will then retract it that far back into the tube; I didn't think this has anything to do with mid-print switches?

ghzserg, 2025-09-15:
No reboot needed
I don't understand what it means to unload after the extruder is running.

Auto insertion parameters only work when inserting a rod into the IFS and indicate how much to tighten it. The extruder is not used there.

ninjamida, 2025-09-15:
> No reboot needed
Thanks for confirming. If I change them mid-print, does it take effect immediately or only after the print is finished?

>I don't understand what it means to unload after the extruder is running.
My impression was that (on native screen), during a color change, the flow of events is as follows:
1. The filament is cut
2. The print head moves to the waste chute
3. The extruder unloads the filament until the filament is outside of the extruder
4. The IFS then unloads the filament further <--- **This is the part I was wondering how to customize**

But based on your last comment, maybe my impression is wrong, and they just work together as needed - if so, all I need to do is increase filament_unload_after_cutting. It does seem unexpected that the unload would happen before moving to the waste basket (on native screen, it moves to the waste basket first, then unloads) but this isn't important; it should still work correctly either way.

> Auto insertion parameters only work when inserting a rod into the IFS and indicate how much to tighten it. The extruder is not used there.
Yes, that's what I thought.

ninjamida, 2025-09-15:
> This is used here `_IFS_REMOVE_PRUTOK`
> 
> ```
> G1 E-{nozzle_cleaning_length} F{filament_unload_speed}
> IFS_F11 PRUTOK={prutok} LEN={nozzle_cleaning_length*1.67|int} SPEED={filament_unload_speed*2} #WAIT=0
> ```
> 
> Extruder removes nozzle_cleaning_length IFS removes nozzle_cleaning_length*1.67 All this happens at the same time

I just found that too; I was looking at the code that uses nozzle_cleaning_length to see if it might be what I'm after, simply because it seemed to be the only parameter that had a default value in the right ballpark for what I was after.

So, I think then - there's no issue (at least so far - still need to test) with zmod itself, but rather, an issue with the documentation:

` nozzle_cleaning_length — Filament pull length for nozzle cleaning when a spool is no longer used. Default: 60 mm. `

To me, nothing about this suggests it's what I'm looking for here. The filament is already outside the nozzle, and cut, at this point; and it doesn't clean the nozzle in any way (everything that's inside the nozzle, stays there; this only removes filament that's not quite inside the nozzle).

I'm kind of bad myself when it comes to thinking of ways to word things like this, but maybe something like "Length to retract filament when unloading during a color change" is better?

I did see you have a dedicated thread for documentation fixes; I'll take a look at it and submit this properly (perhaps after I put a bit more thought into the wording) once I've done some testing on it.

ghzserg, 2025-09-15:
The documentation is originally written in Russian, and then translated into English through a translator. If you can formulate it better, I'm only glad.
Also, as an option, you can not use the value 0.67 - which is obtained from the native firmware, but enter a separate parameter something like nozzle_cleaning_ifs_add_length
Which will indicate how much longer the IFS should work after the extruder stops.
If nozzle_cleaning_length = 60
And nozzle_cleaning_ifs_add_length = 40
Then the Extruder and IFS will extract 60 mm together, and then the IFS will separately extract another 40 mm

ghzserg, 2025-09-15:
It is true that it is worth considering that the extruder pulls at twice the speed that the IFS operates.