ghzserg/zmod #168: Проблема (mainsail): роутинг на frontend

korniux, 2025-09-15:
- [X] У меня AD5M
- [ ] У меня AD5M PRO
- [ ] У меня AD5X

- [] Я работаю с родным экраном
- [ ] Я работаю без экрана
- [X] Я работаю с Guppyscreen

- [X] Я работаю с родным klipper
- [ ] Я работаю с klipper 13
(замените пробел на x внутри скобок)

----

Mainsail использует фронтовый роутинг (History API) и при обновлении страницы через браузер - 404. На nginx решается через `try_files $uri /index.html`.

За fluidd сказать не могу, но думаю стоит проверить.

ghzserg, 2025-09-15:
Используйте Ctrl + F5 
Или опишите подробнее что у вас не работает.

ghzserg, 2025-09-15:
В моде используется httpd из BusyBox и в нем насколько помню нет такого функционала. Можно только 404 ошибку переопределить, но это не совсем чистое решение

korniux, 2025-09-15:
*upd. допллнил шаги воспроизведения*

в httpd есть [FallbackResource](https://httpd.apache.org/docs/trunk/rewrite/remapping.html#fallback-resource) при наличии модуля `mod_dir.c`

ghzserg, 2025-09-15:
Вы путаете Apache httpd и BusyBox httpd.
Ни nginx ни апач не влезет на ad5m

ghzserg, 2025-09-16:
В версии 
1.6.1-30-gc55c25d8 для AD5M реализован свой веб сервер

closed 2025-09-16 by korniux