ghzserg/zmod #174: Problem Nozzle Not Heating Correctly During IFS Filament Change

not-crazyramen, 2025-09-17:
- [x ] I have AD5X

- [ x] I work with Guppyscreen

Detailed error description:

During a multimaterial print, when the filaments change, the nozzle first heats to the default nozzle temperature for the material, then cleans the nozzle, then heats to the correct nozzle temperature from the slicer while still next to the rubber wiper. 

This leads to oozing in the time it takes to go from 220 -> 230 (my slicer setting for the print temp for this PLA+).

It should heat to the temperature set from the slicer before it cleans. 

I've been trying to track down where this is coming from, but I'm not a programmer and have hit a wall.
Let me know if any logs are needed.

ninjamida, 2025-09-17:
Another thought: It should be relatively trivial to modify _INSERT_PRUTOK_IFS so that if a print is in progress, it only uses the extruder's current temperature, instead of the value from filament.json.

You could then add some custom GCode in your slicer in the "Filament Change" section, so that the temperature change gets applied (via M104, not M140) before the filament change begins. As long as the temperature is still high enough to extrude the material (even if messily), it should mean that any defects occur in the poops, not in the print - and defects in the poops isn't a huge problem.

ghzserg, 2025-09-17:
Changing filament works like this.
The filament temperatures are loaded from filament.json.
The mod can't rely on the slicer, as the T1 filament change command may occur before the new filament temperature is specified. Also, the new filament temperature may not be specified at all through the slicer. All of this is configured by the user.
Therefore, when removing the current filament, it checks that the temperature is not lower than the specified one for the current filament, and then heats up to the new filament change temperature.
Adjust the temperature values ​​in the file according to the temperatures at which you print.

closed 2025-09-19 by ghzserg